pflarue/ardop #45: Create a test plan

pflarue, 2024-05-30:
The Ardop specification says: "to insure interoperability between implementations it is required that each implementation that
identifies itself as “ARDOP compatible or compliant” successfully complete and document a compatibility/conformance test that covers the primary operating modes of the protocol."  However, I don't believe that anyone ever documented what such a test would require the implementation to demonstrate.

I can imagine two main parts to such a test.  The first would verify that audio produced by an implementation for the various frame types is as described in the specification, and that the implementation correctly decodes received audio that matches that description.  I believe that while it might be possible to identify failures for this portion of the testing, success might be much harder to define.  This portion of the testing is not what I propose to develop a plan for via this Issue.

The other portion of the conformance testing would focus on the FEC and ARQ protocols and on the Host interface.  Unlike testing of audio modulation and demodulation, I believe that it should be significantly easier to define pass/fail tests for these elements.  I believe that a series of tests could be defined in terms of:
1. An initial state
2. An input via the host interface OR from a decoded received frame
3. A resulting action that changes state variables, sends a message to a host interface, and/or transmits an appropriate frame.
Some tests may also involve time dependent actions such as waiting a certain amount of time for some result or repeating some action a specified number of times at specified intervals.  

For this test plan, I assume that all modulation/demodulation and sending/receiving of audio would be simulated.  Thus, a test plan must consist of (1) a list of the conditions, inputs, and results as described above.   It must also (2) describe how to implement the tests by establishing the initial states, simulating the required inputs, and evaluating whether the correct responses occurred.

I believe that (2), as defined in the previous paragraph, is largely about choosing and applying a unit test framework as described in Issue #30, while (1) is focused on what high level behaviors that we are trying to verify.  These high level behaviors, in conjunction with a study of how ardopcf is implemented, may help define tests for lower level implementation specific behaviors whose testing is also useful.

cbs228, 2024-06-01:
> (1) is focused on what high level behaviors that we are trying to verify.

This sounds to me like an RFC-style spec document. (And maybe draw the ARQ state machine.) I agree that the existing ARDOP spec is probably not enough for a full-fledged implementation with all the details. Better specs and testing against some reference implementation might improve interoperability… but hams have limited developer-hours and a lot of ideas. Interoperability is not always an overriding concern.

If you do write additional documentation, I strongly recommend using markdown or plain text, with SVGs for any vector graphics. Word documents can't be diffed or merged, which make them the bane of my existence.

In the absence of spec, I would probably just write up commented test-cases that describe what they are testing, i.e.: "verify that a receive timeout triggers a DISConnect."

pflarue, 2024-06-01:
The style and format of what I am proposing doesn't matter.  Plain text is probably best.  I agree that nothing technical in nature whose substance is more important than it's format should be written with a word processor.  

A list of statements like your example of "verify that a receive timeout triggers a DISConnect." is what I am looking for, though most tests will require a few additional details to define the initial state or the input that should trigger the test.  Thus, an outline format might be useful so that a common initial conditions can be described for a series of tests for reactions to various inputs.

What is important is that such a list of tests be constructed from a reading of the specification rather than by examining the source code.  If tests are written only to confirm that functions in ardopcf do what they appear to do (by choosing what to test by examining the source code), then errors in the code might be mistakenly identified as correct behavior.