pflarue/ardop #54: decode error

jagdkampf, 2024-06-09:
This occured while monitoring the traffic between 2 other stations. My station is running ardopcf 2.0.1.2 and was in listen mode.  During the
transaction between the 2 other stations, my ardop gui initially read the receive frame as FEC 200.100.50 but switched to FEC 2000.600.60 when the file transfer began even though the signal bandwidth in the waterfall did not exceed 500 hz.  When the decode changed  to 2000 in the receive frame my garim and ardopcf froze and crashed and required a restart of the rigctld and ardopcf and garim to regain operation again.  I will attach the debug log.
[ARDOPDebug8515_20240608.log](https://github.com/user-attachments/files/15749975/ARDOPDebug8515_20240608.log)

pflarue, 2024-06-09:
When ardop is listening for incoming traffic, it sometimes mistakes non-ardop traffic for ardop traffic.  I suspect that this is what you were seeing. This is normal behavior, and not a problem.

However, in some cases, this causes ardop to crash.  This is a problem.  I've not yet been able to identify where in the code the crash is occuring.  The best technique I've found to duplicate the problem is to tune to the FT8 frequency on an active band where there is always plenty of strong FSK modulated signals to confuse ardop, and to turn the incoming audio level up high.  This causes lots of misidentified ardop traffic, and occasionally one causes a crash.  

I first observed this while doing some testing of the waterfall and spectrum displays for the Webgui.  The FT8 frequency was someplace I knew there would always be a signal to observe.

If anybody happens to get a recording of a signal that causes the crash, that would be useful for debugging purposes.

digital-night, 2024-06-09:
Hi Peter, this might be a huge ask, but I'm wondering. Would it be possible to enable a circular audio log of the last minute or so, making it possible for us to submit to you the actual data that caused the crash?

73

Bill Buhler 
AF7SJ

pflarue, 2024-06-09:
@digital-night,

I've thought a little about a similar idea using a circular buffer to allow recording to be started after something interesting is detected, but haven't tried to implement it.

Unfortunately, it might not be as useful for this problem or any other that causes a crash because failure to flush and/or close the file before the crash occurs would be likely to lose the final intetesting parts.  Flushing the write buffer after every write, which might avoid this problem, could seriously degrade performance.  

My idea for trying to debug this crash cobdition, which I haven't tried yet, is to make some extended recordings of FT8 traffic while not running ardopcf.  Then, try feeding them to ardopcf using the --decodewav option to see if I can cause such a crash.  If I find one, then I could create snippets from that audio file until I identified a relatively short segment that would repeatably cause a crash.

digital-night, 2024-06-09:
@pflarue that makes sense, so let me ask a crazy follow up question. How hard would it be to setup the audio in subsystem to copy the sound input to an optionally opened TCP port? Then someone could build a listener that would open the port and receive the data as it came in. Being a separate process the listener process shouldn't crash if ardopcf does and if the socket stopped sending data it could dump the buffer to a file assuming a crash occured.  Later on that same secondary daemon could be expanded to implement your recording of stuff that looks interesting.

pflarue, 2024-06-09:
@digital-night,

All of that might be possible, and some variant of it might even be interesting to explore.  However, at least for me, it wouldn't be quick or easy.  I'm busy enough right now not to want to go that direction.

pflarue, 2024-07-01:
As mentioned in my closing comment to Issue #37, using ASAN/UBSAN I think that I've recreated this crash condition.  While I don't yet fully understand the source of the problem, I have some additional details that will hopefully help me to figure it out and fix it.

pflarue, 2024-07-06:
PR #74 fixes a bug could cause ardopcf to crash while attempting to demodulate/decode invalid or very noisy data.  I believe that bug is what caused the crash reported in this Issues.

closed 2024-07-06 by pflarue