pflarue/ardop #71: Overhaul or replace logging framework

cbs228, 2024-07-05:
(Continued from discussion in [#50](https://github.com/pflarue/ardop/pull/50#pullrequestreview-2159370820))

The existing console and file logging framework has some issues:

* It is part of the platform-specific code, which may introduce unwanted platform-specific differences.

* It uses unsafe `sprintf()` and `strcat()` functions. These need to be replaced.

If you don't mind the logs looking a bit different, it may be more expeditious and safer to replace the logger with something like [zf_log](https://github.com/wonder-mice/zf_log) or [rxi/log.c](https://github.com/rxi/log.c). Of these, zf_log looks much more mature. Both are pure C, require no build system, and are MIT-licensed. There are undoubtedly many others.

pflarue, 2024-07-28:
As I wrote elsewhere, until some of the existing known issues are resolved, I'm not interested in spending a lot of my time fixing things that don't seem to be broken.  That said, if you want to overhaul the logging system, you are welcome create a pull request to implement the necessary changes.  I am OK with using a logging implementation added as a new item in the lib directory or with organic changes to the existing implementation.  Rather than trying to evaluate the alternatives you propose, I suggest that you choose the approach/library that you prefer.  You should confirm that what you choose to do is compatible with the existing usage.  The following are some factors you should probably consider.

* Compatible license.
* Compatible with our build system.
* Compatible with both Linux and Windows.
* Will both produce log files and write log statements to console as appropriate for the specified log level.
* Log files must be written to the directory indicated by the --logdir option or the current directory if that option is not used.  Log filenames may differ from those currently used, but should probable begin with ARDOP (or some variation thereof), and must include the port number and UTC date.  Filename should not include spaces or other characters that require escaping or quotes around them when referenced on the command line.
* Logging must respond to changes from LOGLEVEL and CONSOLELOG host commands, which are independent of each other and which may occur at any time.
* Permissible log levels must accommodate both the LOGDEBUG and (the nonstandard but useful) LOGDEBUGPLUS levels currently is use.  Ideally, the range of log levels should be extensible.
* Messages written to the log file must include millisecond resolution UTC timestamp values.
* Currently, log messages are created with either WriteDebugLog() and Debugprintf().  I'm OK with the function name changing, but changing the arguments in any way that cannot be automated with search/replace is probably not a good idea.  I'm OK with eliminating DebugPrintf, in favor of WriteDebugLog() with an elevated log level if you'd prefer.
* Webgui.c includes the definition of callback functions passed to ws_set_error() and ws_set_debug().  I don't mind having to do some rewrites to these to accommodate changes to the logging interface. 

If you have additional questions, ask them here.  I don't intend to do a detailed code review for a pull request addressing this issue.  Instead, I will compile, run any related tests you choose to create, and then run ardopcf to compare log output with results produced prior to the changes.  I'll do this on each of the supported build platforms.

pflarue, 2024-09-27:
This was done by PR #88, which has been merged into the develop branch.  The changes will be merged into the master branch when the next release of ardopcf is produced.

closed 2024-09-27 by pflarue