pflarue/ardop #96: "CQ" callsign and ARQ sessions

cbs228, 2024-09-26:
As I work #79, I am taking a closer look at callsign processing within ardopcf.

The latest [TNC interface spec](https://www.winlink.org/sites/default/files/downloads/ardop_tnc_host_mode_interface_spec.pdf) I can find specifies that:

> `ARQCALL CQ 5`

is valid and is a connection request to "any" station. Legacy ARDOP appears to support this syntax on the sending side, but there are no such checks on the receiving side. The receive logic does not answer CQ. This makes sense for several reasons:

* The called station appears to use whatever `MYCALL` or `MYAUX` was dialed as its final `IDFrame`. CQ is not valid for ID frames.

* Generating a session ID requires knowledge of the target station's callsign.

  * The target's proper MYCALL/MYAUX callsign is not reported back in the `ConAck`; the dialing station has to know it.

  * We could use `CQ` as the target callsign, but multiple stations may answer the connect request and will have the same session ID. This will cause collisions.

* The TNC specification does not allow `CQ` to be assigned to any of the `MYAUX`—probably for the above reasons.

* There is no way to request a particular application-layer protocol, so who knows what that might be. You might reach a WL2k node or a live keyboard operator; there's no telling.

* For WL2k in particular, US amateur radio stations have to abide by certain third-party country relay restrictions. This makes it important to know which station you are `ARQCALL`ing.

All of this suggests that a `ARQCALL CQ` can't actually work within the current protocol. Are we aware of any implementations of it? If not, I will make no special attempt to support it.

It would be nifty to be able to `PING CQ`, or a group/tactical callsign, but that would *definitely* require protocol revisions. The `PingAck` would need to include the sending station's callsign (#95), plus some way to mitigate an "ack storm" of multiple stations trying to reply at once.

This is mostly just a request for information. Unless there is a serious appetite for spec revisions, we should probably just close this.

pflarue, 2024-09-27:
I had noticed a couple of references to CQ in the code, but had not explored their significance.  I suggest that you send a message to the Ardop users group about this to see whether any users know of how this is/was used or intended to be used.  You might also get a response from one of the people who worked on development of the Ardop specification.

I believe that a more appropriate way to "call CQ" with Ardop, if one wanted to do so, would be to send a FEC data frame with a CQ, the caller's Callsign, and whether they are looking for an FEC data frame or an ARQ ConnectRequest as a response.  I can imagine this being done as an unstructured human readable FEC data frame, or being encoded into a host level protocol.

The suggestion of a PING CQ is probably also something better implemented via an FEC based host level protocol rather than by using Ardop's Ping and PingAck frames.  Such an implementation might reduce the "ack storm" effects by having responding stations each choose a random delay before sending their Ack (also as an FEC data frame) and by using additional random delays if the busy detector indicated that the channel was not clear.

Lets leave this issue open for a while to see if anyone else responds.

pflarue, 2024-10-07:
PR #99 eliminates all code specific to handling of CQ as a StationID.  As mentioned in the PR comments, there is a dicsussion related to use of CQ with Ardop at the ardop users group:
https://ardop.groups.io/g/users/topic/use_of_cq_with_ardop/108851759

closed 2024-10-07 by pflarue