pflarue/ardop #97: Failure in test/ardop/test_log.c (Windows)

pflarue, 2024-10-03:
I'm seeing the following when running `mingw32-make test` on Windows (both 32bit and 64bit mingw):
```
...
[ RUN      ] test_ardop_logfile_handle
[  ERROR   ] --- test/ardop/test_log.c:26 error: Type mismatch: name[NULL] expected[FILE*]in __wrap_fopen
...
```

A second identical error is produced for `test_ardop_logfile_write`.

Though these tests are failing, log files are being written correctly during ardopcf use.

I know that I tested PR #88 on Windows before merging it, but it is possible that I failed to run the final version of these tests (or else I would have encountered this error then.)

I've been using my Linux machines to work on some new fixes, and found this error upon switching to Windows to try the new changes there.

@cbs228 can you offer any suggestions?

cbs228, 2024-10-04:
I would definitely try a `make clean` first, if you haven't already.

Does the same failure happen on 32-bit vs 64-bit builds?

Can you get me a stack trace and local variable dump of where the problem occurs? To get a useful stack trace, you will need to add a breakpoint on the `_fail` function, which is part of cmocka. See the example gdb session below.

The failure message suggests that our mocked `fopen()` is receiving a nullptr for its path name. This shouldn't be possible within `ardop_logfile_handle()`, but perhaps something very strange is happening.

<details>

<summary>Example gdb session</summary>

I caused this failure by adding a bogus assert.

```log
$ gdb test/ardop/test_log
Reading symbols from test/ardop/test_log...
(gdb) b _fail
Function "_fail" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 1 (_fail) pending.
(gdb) r
Starting program: /build/test/ardop/test_log 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[==========] tests: Running 5 test(s).
[ RUN      ] test_ardop_logfile_calc_filename
[       OK ] test_ardop_logfile_calc_filename
[ RUN      ] test_ardop_logfile_need_rollover
[       OK ] test_ardop_logfile_need_rollover
[ RUN      ] test_ardop_logfile_handle

Breakpoint 1, 0x00007ffff7f8a178 in _fail ()
   from /lib/x86_64-linux-gnu/libcmocka.so.0
(gdb) bt
#0  0x00007ffff7f8a178 in _fail () from /lib/x86_64-linux-gnu/libcmocka.so.0
#1  0x00007ffff7f8a253 in _assert_true ()
   from /lib/x86_64-linux-gnu/libcmocka.so.0
#2  0x0000555555555aa7 in test_ardop_logfile_handle (state=0x55555555d310)
    at test/ardop/test_log.c:130
#3  0x00007ffff7f8cfdf in ?? () from /lib/x86_64-linux-gnu/libcmocka.so.0
#4  0x00007ffff7f8d755 in _cmocka_run_group_tests ()
   from /lib/x86_64-linux-gnu/libcmocka.so.0
#5  0x0000555555556260 in main () at test/ardop/test_log.c:239
(gdb) f 2
(gdb)       # NOTE: I selected the stack frame that contains
(gdb)       # the unit test code
#2  0x0000555555555aa7 in test_ardop_logfile_handle (state=0x55555555d310)
    at test/ardop/test_log.c:130
130		assert_false(true);
(gdb) info locals
UNIX_TIME = 1722724636
UNIX_TIME_LATER = 1722767836
uut = {logdir = 0x555555559044 "", stem = 0x5555555591b8 "ARDOPDebug", 
  extension = 0x5555555591b3 ".log", port = 0, opened_at = -2147483648, 
  handle = 0x0}
out = 0x0
```
</details>

pflarue, 2024-10-06:
@cbs228 Thanks for helping me understand the error message.  PR #98 fixes the problem.

closed 2024-10-06 by pflarue