pflarue/ardop #108: Please document command line options

radio-miskovice, 2024-10-11:
Command line options definition (option, usage, what they do) is nowhere to find. IMHO it should be in the README.md or, if too long, specified in a spearate document referred by hyperlink from README.

pflarue, 2024-10-11:
Jindřich,
Thank you for this.  I recognize that a lot of work is needed to improve documentation, and this needs to be part of it.

For now, typing `ardopcf -h` or `ardopcf --help` shows a summary of command line options.

radio-miskovice, 2024-10-11:
Hi, if you allow me to publish branch, I will push a new branch with the markdown file in docs. I am almost done :)

pflarue, 2024-10-11:
Please create a pull request.

For code changes, pull requests should be to the develop branch.  However, pull requests for documentation may be to the master branch.

radio-miskovice, 2024-10-12:
I created PR into develop branch, pls have a look.

closed 2024-10-12 by radio-miskovice