pflarue/ardop #111: Possible Buffer Overflow issues

stclairc, 2024-10-17:
When building from source the following warnings are observed:
HostInterface.c:1099:48: warning: ‘%s’ directive writing up to 2099 bytes into a region of size 88 [-Wformat-overflow=]
HostInterface.c:1005:48: warning: ‘%s’ directive writing up to 2099 bytes into a region of size 88 [-Wformat-overflow=]
HostInterface.c:247:48: warning: ‘%s’ directive writing up to 2099 bytes into a region of size 88 [-Wformat-overflow=]
SoundInput.c:3413:37: warning: ‘ > ’ directive writing 3 bytes into a region of size between 0 and 31 [-Wformat-overflow=]
../ARDOPCommonCode/ALSASound.c:392:55: warning: ‘%02d’ directive writing between 2 and 5 bytes into a region of size between 0 and 18 [-Wformat-overflow=]

I suggest changing to "snprintf" to help avoid the possible overflows.  
Sizing on cmdCopy and strFault appear insufficient.

pflarue, 2024-10-17:
Can you please indicate which commit you were compiling from to produce these warnings.  The line numbers referenced do not appear to match the current develop branch.

stclairc, 2024-10-17:
Sorry, I had downloaded the Main branch.
And I'm still getting the hang of using GitHub for anything more than simple, one-person projects.

The develop branch shows similar errors:
[Errors.txt](https://github.com/user-attachments/files/17423838/Errors.txt)

pflarue, 2024-10-17:
@stclairc, There is no need to apologize.  I am also still learning how to use GitHub (and git on my own computers).  Working on ardopcf is my first attempt at trying to maintain a project that others are contributing to.

However, the contents of your linked Errors.txt still doesn't seem to match a recent version of the develop branch.  Also, as I wrote before, it would help if you indicate the specific commit that you are compiling from.  The develop branch continues to advance with each new commit that I push or Pull Request that I merge into it.  So, for example, as I am writing this, the most recent commit in the develop branch is 0990a99 resulting from merging PR #112.  Before that, the most recent commit in the develop branch was b96c6eb "Minor enhancements to the Webgui."

Assuming that you are using git from the command line on your computer after cloning this repository, try `git log --all --graph --oneline` to see the sequence of commits and their relationships, listed starting with the most recent.