pflarue/ardop #113: WIndows 10: no output from USB sound card (Kenwood TS-590)

radio-miskovice, 2024-10-18:
Hi, not sure if this is issue of the Ardop software, but it could be related to device handling and selection on Windows.

When I start ARDOPCF on windows and set all the correct devices, the transceiver has no output.
PTT and CAT works correctly.
When I select PC speakers as the audio output device, I do hear the signal when packet is sent, from speakers.
I have double-checked TS-590 setup: audio input is selected from USB line and the transceiver transmits other modes from the same PC (JS8 etc.).
I can see one strange thing in Windows configuration: when I open Volume Control panel, I can see ARDOPCF as signal source both in PC Speakers and the USB Audio Device (TS-590) at the same time, which to me suggests there might be something wrong with output audio device initialization.
I am not asking for fix, but rather would like to ask for guidance/reference in which part of the software to look for possible fix. I would like to compile the program for myself in Visual Studio, if possible, because this is what I have available.

pflarue, 2024-10-18:
@radio-miskovice,  your description of the problem is strange.  I tried ardopcf (both from the master branch and the current develop branch) on a Windows 10 machine and was unable to duplicate your issue.  I tried both a Digirig sound device and a generic USB soundcard so that I had a second sound device similar to what you have when connected to your TS-590.  I have never used a TS-590, but I assume that the computer interacts with it like any other USB sound device.  If you have any other USB sound device that you could try, that might help confirm that the problem is not specific to the TS-590 and its configuration.  It is very strange that it works correctly with JS8 but not ardopcf.

Your description of seeing the ARDOPCF as signal source in both PC Speakers and USB Audio Device is very strange.  Do you have any virtual audio cable software installed that might be creating a link between these devices?

I assume that you have checked all of the audio level settings in the Windows Sound controls for the speaker device from your TS-590 to ensure that the audio level is not too low, muted, or has the balance set all the way either left or right.  The "Disable all Enhancements" box should also be checked and "Spatial sound format" should be "Off".  I notice that there are "Preview" buttons on both the "Enhancements" and "Advanced" tabs on this Speaker Properties dialog.  If the TS-590 does not receive audio transmitted by ardopcf, see whether it receives audio produced by pressing one of those "Preview" buttons.  I assume that you have used the buttons on the ardopcf WebGui for "Send2Tone" and/or "SendID" to send test output to your radio.  

I notice that changing which devices are defined as the "Default Device" and "Default Communications Device" changes the device number assigned to the built-in speaker/microphone and the USB speaker/microphone.  If such changes are made while ardopcf is running, it may be necessary to stop and restart ardopcf with the new correct device numbers.  Since plugging/unpluging a USB audio device also changes the device numbering, I wonder whether powering on/off your TS-590 may also change the numbering.  I usually set the built-in speaker/microphone to "Default Device" and the USB sound card to "Default Communications Device".  

Might the problem be related to using the left or right audio channel?  I noticed that my digirig appears to windows as a stereo device, but that if I set the Balance to output only to the right channel no audio is sent to the radio.  So, only left channel audio seems to be used by the digirig for transmit.  While ardopcf has command line options to select use of left or right audio channel, I don't believe that the Windows audio functions currently use these values like the Linux functions do.  I think that ardopcf always sends audio to both channels of a stereo device, but I'd need to look at this further to verify that.  If nothing else works, let me know and I'll explore this possibility further.

You can try compiling ardopcf from the latest develop branch to see if that helps.  I do not believe that ardopcf will compile correctly for Windows using Visual Studio, which you mentioned.  Instead, I believe that you need to use MinGW.  I recommend one of the easily installed packages from https://winlibs.com.  I'm in the process of writing some new documentation for ardopcf including step by step instructions for building from source both for Windows and Linux.  I hope to get this pushed in GitHub today or tomorrow.  I'll provide a link in another comment to this Issue when those instructions are ready.  

If you compile ardopcf from the latest develop branch, and you want to test ardopcf without a host program (like Pat Winlink) connected that will set your callsign, then you will discover that the Send2Tone and SendID buttons in the WebGui do not work.  To fix this use --hostcommands "MYCALL OK4RM" to set your callsign while starting ardopcf.  This is required by a recent change to fix sending of ID Frames containing the station callsign.  A part of that fix prevents ardop from transmitting anything (including the 2tone signal) until a callsign is set.

It may be helpful to send a message about your problem to the users subgroup at ardop.groups.io.  I believe that most ardopcf users are using Linux, but it is possible that a Windows user in that group might be able to help.

None of these suggestions seem very likely to fix your problem, but they are the best that I can think of right now.

pflarue, 2024-10-18:
I also suggest that you try all of the available audio devices.  Maybe the problem is that ardopcf is labeling them incorrectly.

pflarue, 2024-10-20:
I just added [BUILDING.md](https://github.com/pflarue/ardop/blob/master/docs/BUILDING.md) to the master branch.  If you want to try building ardopcf from source, this may be helpful.

I'm still working on additional updates to documentation.  Since this file might help you resolve your issue, I decided to merge it into the master branch now rather than wait for the remaining updates to be ready.

radio-miskovice, 2024-10-21:
Hi, Peter, this looks almost 100% like a Windows issue, not the software. I will try ardopcf on another PC with the same TRX and if it works, I will close this issue. Sorry for the buzz.

radio-miskovice, 2024-10-22:
I did not repeat test with as I had to prepare digirig cables for TX-500 and test. Had a successful winlink session although I think winlink express is pretty lousy and did not work very well with TX-500 (ARDOP only when I dialed frequency manually, VARA session seemingly handled frequency changes but did not manage to transmit; Varac has no problems at all with TX-500). My problem was obviously Windows and nothing else. Also built ardopcf successfully on another windows pc.

closed 2024-10-22 by radio-miskovice