pflarue/ardop #124: ardopcf fails to build with GCC 14.2.1 on Fedora

hobbes1069, 2024-11-29:
While trying to build the project I ran into the following errors:

```
gcc -g -MMD -Isrc -Ilib  -c -o lib/rawhid/rawhid.o lib/rawhid/rawhid.c
lib/rawhid/rawhid.c: In function ‘HID_Read_Block’:
lib/rawhid/rawhid.c:361:20: error: passing argument 1 of ‘read’ makes integer from pointer without a cast [-Wint-conversion]
  361 |         Len = read(CM108Handle, Msg, 64);
      |                    ^~~~~~~~~~~
      |                    |
      |                    hid_device * {aka struct hid_device_ *}
In file included from lib/rawhid/rawhid.c:8:
/usr/include/unistd.h:371:26: note: expected ‘int’ but argument is of type ‘hid_device *’ {aka ‘struct hid_device_ *’}
  371 | extern ssize_t read (int __fd, void *__buf, size_t __nbytes) __wur
      |                      ~~~~^~~~
lib/rawhid/rawhid.c: In function ‘HID_Write_Block’:
lib/rawhid/rawhid.c:414:29: error: passing argument 1 of ‘write’ makes integer from pointer without a cast [-Wint-conversion]
  414 |                 ret = write(CM108Handle, Msg, 64);
      |                             ^~~~~~~~~~~
      |                             |
      |                             hid_device * {aka struct hid_device_ *}
/usr/include/unistd.h:378:27: note: expected ‘int’ but argument is of type ‘hid_device *’ {aka ‘struct hid_device_ *’}
  378 | extern ssize_t write (int __fd, const void *__buf, size_t __n) __wur
      |                       ~~~~^~~~
lib/rawhid/rawhid.c:419:32: error: passing argument 1 of ‘close’ makes integer from pointer without a cast [-Wint-conversion]
  419 |                         close (CM108Handle);
      |                                ^~~~~~~~~~~
      |                                |
      |                                hid_device * {aka struct hid_device_ *}
/usr/include/unistd.h:358:23: note: expected ‘int’ but argument is of type ‘hid_device *’ {aka ‘struct hid_device_ *’}
  358 | extern int close (int __fd);
      |                   ~~~~^~~~
make: *** [<builtin>: lib/rawhid/rawhid.o] Error 1
make: *** Waiting for unfinished jobs....
```

makrohard, 2024-12-15:
I can reproduce this exakt issue here on fedora 41. The binary does work and it does compile fine at a Linux-Mint System.

ScottDIrwin, 2024-12-17:
The same occurs on Debian 13 / GCC 4.2.0-8.

pflarue, 2024-12-25:
@hobbes1069, @makrohard, @ScottDIrwin:

Thanks for pointing out this problem, and confirming that it was not specific to just one system.  I think that I've fixed it with a one line change that I've pushed to the develop branch.  Please confirm that this works now.

I've long seen this issue, but on my system these are Warnings rather than Errors, so that they don't prevent compiling.  Since this is in code related to using a CM108 (which I don't have to test), and it hadn't prevented me from compiling, I was reluctant to make any changes.  However, on further inspection, the solution seemed simple enough.  I don't like ignoring warnings, and I hope to eventually get rid of all (or at least most) of them.  However, doing so is is not high on my list of priorities right now.

hobbes1069, 2024-12-25:
Thanks for taking a look. Distros like Fedora or Debian my enforce build flags that make some warning errors which is likely what happened here.

ScottDIrwin, 2024-12-31:
I was able to make it without issue on Debian Trixie.  Thanks for the quick fix!

pflarue, 2024-12-31:
@ScottDIrwin,

Thanks for confirming that the fix works.  

Roughly a month after the problem was reported doesn't seem particularly quick for a simple one-line fix to a problem that was easy to diagnose.  Sometimes it is nice to have an easy problem to fix.

closed 2024-12-31 by pflarue