pflarue/ardop #125: PPT with -RTS

makrohard, 2024-12-15:
Ardopcf works fine here, when using digirig-hardware. When using AIOC (https://github.com/skuep/AIOC) however, the TRX does not key-up. TRX is Quansheng uv-k5(8) in my case.

I'm using ardopcf with the parameter -p /dev/ttyACM0 which represents the AIOC device. PTT is triggered by webgui or garim.  Audio does work fine with doth digirig and AIOC. The TRX used, does not support CAT.

I suspect, the problem might be related to the +- RTS / DTR settings. 

The AIOC-Device PTT is proven to work work with:
* direwolf (using -RTS)
* freeDV (using +DTR)

Is there any way to configure +/- RTS/DTR settings? 
If not: Is there any way to get ardopcf PTT to work with the AIOC-Device?

pflarue, 2024-12-25:
@makrohard,

Currently using the -p or --ptt option is interpreted to mean use SetRTS to transmit, and then ClearRTS to stop transmitting.  There is also code (for both Linux and Windows) to use SetDTR to transmit, and then ClearDTR to stop transmitting.  However, there is no command line option to indicate that this DTR code should be used.  

I can add this option, but have no way to test whether it is working correctly since I don't have a device that requires DTR.  If I create a new branch with an option to use DTR instead of RTS can you build ardopcf and verify that it works as expected?  Since you are using `/dev/ttyACM0` for the port, it appears that you are using Linux.  It would be great if you or somebody else with this hardware could also test it for Windows.

pflarue, 2024-12-25:
@makrohard, 

Please try the new ptt-dtr branch.  As indicated by the updated help screen in that branch, using -p DTR:/dev/ttyACM0 or --ptt DTR:/dev/ttyACM0 should use DTR instead of RTS to activate PTT.  However, as I mentioned in my earlier comment, I don't have hardware that requires DTR to test this with.  So, let me know if it works as expected.  I believe that this should work for both Linux and Windows.

makrohard, 2024-12-26:
Just tested, and it works like a charm, thank you very much!

My setup: Quangsheny UV-K5(8) + [AIOC Cable](https://github.com/skuep/AIOC)

### Serial adapter
```
$dmesg
hid-generic 0003:1209:7388.0004: hidraw1: USB HID v1.00 Device [AIOC All-In-One-Cable] on usb-0000:00:14.0-2.1/input3
cdc_acm 3-2.1:1.4: ttyACM0: USB ACM device
```

### Sound device
```
$ arecord -l
**** List of CAPTURE Hardware Devices ****
<snip/>
card 1: AllInOneCable [All-In-One-Cable], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

### ardopcf

```
$./ardopcf 8515 plughw:1,0 plughw:1,0 **-p DTR:/dev/ttyACM0** -G "-8514" -H "MYCALL EXAMPLE"
ardopcf Version 1.0.4.1.3_develop (https://www.github.com/pflarue/ardop)
Copyright (c) 2014-2024 Rick Muething, John Wiseman, Peter LaRue
See https://github.com/pflarue/ardop/blob/master/LICENSE for licence details including
  information about authors of external libraries used and their licenses.
LinBPQ Port /dev/ttyACM0 fd 4
Using RTS on port /dev/ttyACM0 for PTT
ARDOPC listening on port 8515
Capture Devices
<snip/>

Card 1, ID `AllInOneCable', name `All-In-One-Cable'
  Device hw:1,0 ID `USB Audio', name `USB Audio', 1 subdevices (1 available)
    1 channel,  sampling rate 8000..48000 Hz

Using Both Channels of soundcard for RX
Using Both Channels of soundcard for TX
Opening Playback Device plughw:1,0 Rate 12000
Opening Capture Device plughw:1,0 Rate 12000
ardopcf listening on port 8515
Webgui available at port 8514.
Setting ProtocolMode to ARQ.
```

ardopcf does key and un-key the TRX quite fine now using the AIOC-Cable
Switching back to the digirig-device, does work fine too, by omiting **:DTR** in the **-p** option.

The  ptt-dtr branch does fully work as expected with my setup.

closed 2024-12-26 by makrohard

pflarue, 2024-12-26:
OK.  I've merged this into the develop branch.  I'm going to post a request to the users subgroup of ardop.groups.io to see if there is somebody in that group that has a windows machine and a device that requires DTR for PTT and can test this change for windows.

makrohard, 2024-12-31:
We did some installations of ardopcf (+ (g)arim) on 38C3. PTT is always tricky, but we managed to get 'most' TRXes running. Nobody got a windows machine, there :-)