pflarue/ardop #130: MacOS place holder

edfardos, 2025-01-01:
Great work on this guys - really.

I just cloned this to MacOS to inventory what's available and what compiles.

ALSA is linux specific, and could be replace by PortAudio (cross platform audio library).

librt is missing
rawhid is missing
libasound missing (replace with portaudio possibly)

and, after commenting out those deps we get to this,

```
craiger@Craigs-Mac-mini ardop % make
gcc -Xlinker src/common/ardopcf.o -o ardopcf  -lm -lpthread
Undefined symbols for architecture arm64:
  "_platform_main", referenced from:
      _main in ardopcf.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

pflarue, 2025-01-01:
I would very much like to see ardopcf become usable on MacOS.  Since I neither have nor expect to get a Mac, I would welcome help from someone willing to work on this.  Ideally, that help would include writing and testing the MacOS specific code, writing MacOS specific documentation, and providing binaries for future releases.  While the documentation and binaries would be nice to have, the writing and testing of new code is obviously the first step.

I am currently working to improve the startup process, including better error checking and better error messages when a problem occurs during startup.  As a part of this, I am moving as much code that was previously duplicated in the Windows and Linux specific files into ARDOPCommon.c.  I also plan to separate audio system specific functions and the other OS specific functions into different files.  

While these changes should improve usability and maintainability, I think that they will also provide a better starting point for adding native support for other audio systems, and perhaps other operating systems.  I hope to push these changes to GitHub within the next few days.  The OS-specific platform_main() functions will be simpler when I'm done than they are now, which should make it easier to write such a function for MacOS.

My hope is that since MacOS and Linux are both Unix derivatives, most of the common and Linux code should work as is or with minimal changes.  As you wrote, a new sound interface would definitely be required since MacOS has neither ALSA nor the Windows sound system.  However, it might be possible to implement an additional sound interface that could be used by both the Linux and MacOS builds.  

ALSA is the only sound system available by default on some Linux systems, including the Raspberry Pi Zero 2 W using the Lite version of the Raspberry Pi OS that I mostly use.  So, ALSA support will be retained.  I recently learned that on Linux systems with PulseAudio, the ALSA functions in ardopcf still work.  Thus, I've had no motivation to add native support for an additional sound system.  However, the possibility that ardopcf could run on MacOS justifies exploring this further.

pflarue, 2025-01-19:
It took longer than expected because I also did a major upgrade to CAT/PTT in the process, but I have now merged a change into the develop branch that better isolates the audio system specific code and other OS specific code.  The other OS specific code is mostly PTT related, but also includes some time/date related functions.  I think that these changes should make it easier to add support for MacOS and/or other audio systems.  Since Pat supports MacOS, and thus presumably can handle CAT control including PTT, initial support for MacOS could exclude PTT control.

pflarue, 2025-08-15:
@ccallahan, I'm a bit confused.  This issue relates to adding support for MacOS, which does not yet work.  Are you attempting to use MacOS, or did you respond to the wrong issue?  If you are trying to use MacOS, please wait for further progress.  

If you are not trying to use MacOS, what system (Linux or Windows) are you using, and what version of ardopcf (released binaries, or have you compiled from source using the develop branch) are you using?  Have you looked at docs/USAGE_linux.md or docs/USAGE_windows.md.  The versions of these in the master branch describe usage of the latest released binaries, while the versions of these in the develop branch describe usage if you have compiled from source using the develop branch.

ccallahan, 2025-08-16:
Sorry for the confusion, I was testing @askpatrickw's fork and ran into said issues.

pflarue, 2025-08-17:
@ccallahan.  OK.  In that case I can't help you since I don't currently have a Mac.  Maybe @askpatrickw will respond to your question (or maybe not).

However, unless you might be interested in helping him to rework his fork to target the develop branch, I'd encourage you to be patient.  I'm excited about the potential for ardopcf to support MacOS in addition to Windows and Linux.  However, since I'm not the one doing the work, I'm content to wait for it to happen.  At some point I'm sure we'll be looking for MacOS users to test this new build with various radios and radio interfaces, at which point your feedback will definitely be welcome.

askpatrickw, 2025-08-28:
@ccallahan  if you want to try it ...  using the GH CLI. 

`gh pr checkout 168`

I consider this a draft PR... I think the bones are all there, but it needs more testing.

ccallahan, 2025-09-14:
@askpatrickw I'm currently otw to active POTA, I'll text it while I'm there!