pflarue/ardop #131: Enhancement/Feature Request: RX Level Adjustment

drexjj, 2025-01-08:
I kindly request that you consider adding a RX level adjustment feature to the software which will allow broader compatibility with the sBitx transceiver and other audio devices. 

The sBitx transceiver does not have options to increase RX audio gain level to work with ARDOP, which significantly impacts the audio quality and stability of the ARDOP connection. The audio level is too low in ARDOP for the sBitx resulting in inconsistent audio reception and connection issues. 

Implementation Suggestion:
Add a RX level adjustment parameter to the ARDOP settings, allowing users to fine-tune the RX gain to match their transceiver's settings.

I believe this addition would greatly benefit the ARDOP user community and improve the overall user experience. If you require further information or clarification, please let me know.

pflarue, 2025-01-08:
Unfortunately, I don't believe that any RX audio level adjustment that could be implemented within ardopcf would be helpful.  Ardop should work well with reasonably low audio amplitude so long as the signal to noise ratio is OK.  The audio level indicator in the WebGui is somewhat misleading for low signals because it is a linear rather than log or dB scale.  I should probably change that.  When I created the WebGui, I was more concerned about audio being too high than too low because a signal that "sounds" OK, can be overdriven such that distortion due to clipping can hurt performance.  So, some visual indicator is necessary to inform the user when this is happening.  Of course, if your audio is very very low, then your signal to noise ratio cannot be ok, because the quantization noise power level becomes large compared to the signal power level.  If this is the case, amplifying the audio (which is effectively using only a small part of the 16-bit available range) will amplify the noise as well as the signal, and will not help.  

If your signal to noise ratio is OK, but ardop is not detecting the signal, it may help to try adjusting the squelch level.  Squelch can have a value between 1 and 10, and has a default of 5.  This can be set the the SQUELCH host command, and thus can be set using the --hostcommands (or -H) command line option.  Try --hostcommands "SQUELCH 1" to see if that helps.  Squelch is only used as a part of detecting whether a signal is present to try to decode.  So, if ardopcf is detecting received frames, but then failing to correctly decode them, this will not help.  In the WebGui, frames that are detected, but now correctly decoded means that you will see a frame type highlighted in yellow in the `RX Frame Type:` field, which will then turn red meaning it could not be correctly decoded.  In this case, low `Quality` values and high `RS Errors` values are also displayed below the waterfall plot.  If the signals are not being detected, and thus adjusting the squelch setting might help, then the `RX Frame Type:` field remains empty when you know that you are receiving an ardop signal.

If adjusting the squelch does not help, you need to find a way to increase the audio level before it gets to ardopcf.  Excuse my ignorance about the sBitx.  Everything that I know about it I learned just now after reading your message.  However, if I understand correctly, it is a combination of a software defined radio and a raspberry pi.  So, I assume that the audio level controls typically available on a raspberry pi are still there.  I usually run ardopcf on a raspberry pi without a graphical desktop interface.  So, I use the `alsamixer` command to adjust the receive audio level.  I know that there is something equivalent available on the desktop, but I don't have a pi running a GUI with me right now to see what it is called.  Another way that you might be able to increase the receive audio level is by adjusting the RF or IF gain, or adjusting the AGC settings of the radio.

If none of these help, let me know and I'll try to help you figure out how to get ardopcf working better.  It would help me to understand your problem to know what the "Input peaks" values are when you are receiving an ardop signal.  If you are using the default LOGLEVEL settings, this should be written to the log file about every 10 seconds.  I'm also curious what the WebGui waterfall display looks like when you are receiving an ardop signal.  This doesn't indicate how loud the incoming audio is, but it provides a rough gauge of how the signal level compares to the level of the background noise.

pflarue, 2025-01-19:
@drexjj,

Were you able to get your system working adequately?

closed 2025-03-16 by pflarue