pflarue/ardop #139: Host command CAPTUREDEVICES returns only one device, does not return device description

radio-miskovice, 2025-03-03:
Maybe I misunderstood the logic of CAPTUREDEVICES host command, please correct me if I am wrong. The same applies to PLAYBACKDEVICES accordingly.

What I would expect as response to this host command would be the list of all audio capture devices available in the given system, and maybe this is what it was originally intended. What seems not to have been planned is that the list should include device description as well. However, without description the list itself could not be used reliably to select another device during the same TCP host session. Example: QMX audio interface contains the string QMX which might be helpful if the frontend program presented all audio input devices to the operator and let them pick the one they need.

My preliminary analysis of the problem:

The response to   CAPTUREDEVICES  is to send the string from char* CaptureDevices. (common/HostInterface.c:475)
CaptureDevices is declared in linux/ALSASound.c:101 and in windows/Waveout.c:67 as char* and as such it does not seem to represent a list of anything.

IMHO if we want to see a list of all devices, from the linux part variables `char ** ReadDevices` (linux/ALSASound.c:630) and `int ReadDeviceCount` (linux/ALSASound.c:631) should be used. In the windows part there seem to be similar variables `char CaptureNames[16][MAXPNAMELEN + 2]= {""};` (windows/Waveout.c:77) and `int CaptureCount` (windows/Waveout.c:70). The declarations seem to be a little bit messy in both variants (placed at different places in the code) and that's why it is not easy to find them.

IMHO what CAPTUREDEVICES host command should do is to call `GetInputDeviceCollection()` (linux/ALSASound.c:811) or `GetSoundDevices()` (windows/Waveout.c:565) and present a fresh list. Similarly with PLAYBACKDEVICES.

Finally, my personal opinion: not only is the code and function naming in the linux and wondows parts inconsistent and prehaps confusing, but to me it seems that the whole ALSASound.c is a little bit messy as it handles a number of other things unrelated to sound devices, e.g. some GPIO interaction, serial port, etc.

I am willing to help with the sound device part fix if we agree on a general approach and you give me OK to do that.

pflarue, 2025-03-03:
Currently, the CAPTUREDEVICES, CAPTURE, PLAYBACKDEVICES, and PLAYBACK commands are all broken.  Fixing CAPTUREDEVICES and PLAYBACKDEVICES would be relatively easy.  Fixing CAPTURE and PLAYBACK is more complicated, but until these are also fixed, the other two are of limited use.  The refactoring of code to open and configure ALSA devices that I am currently working on will make it easier to implement CAPTURE and PLAYBACK, as well as CODEC.  A fundamental feature of these changes that I am working on is to open the audio devices and keep them open rather than close and reopen these devices when toggling between RX and TX.

However, these changes are proving more complicated than I hoped they would be.  This is partly because I am working to ensure that the changes also more consistently handle pulse as an ALSA device and the ALSA devices derived from plugins such as dmix, dsnoop, and plughw.  With these changes, use of devices defined in .asoundrc to combine dmix and dsnoop with plughw should allow ardopcf to better share audio devices with other software when the pulse device is not available.  I think that I'm close to getting this right, but I'm not quite there yet.  I'm finding and fixing bugs found by testing the changes on systems with and without pulse, with and without additional items defined in .asoundrc, and with and without various other software using the same audio devices.  Testing all of these variations is inconvenient and takes time, but has proven very useful.

I agree that ALSASound.c and Waveout.c were very messy because they contained both sound system related code and other OS specific code.  Therefore, Pull Request #135 which has been merged into the develop branch replaced them with ALSA.c, Waveform.c, and a version of os_util.c specific to each operating system.  I think that you will like what I did with these as shown in the current develop branch.

I agree that the Linux and Windows functions that write info about available audio devices to the log need to be reworked to have the same signatures for use with CAPTUREDEVICES and PLAYBACKDEVICES.  It may also prove useful to change the amount and format of text produced.  Before doing this, I think that the Linux version should be expanded.  Currently, the lists produced are similar to using the `--list-devices` option with `aplay` and `arecord`.  Instead, I think that it should be similar to using the `--list-pcms` option.  The `hw:0,0` currently shown is not usually useful because most hardware doesn't natively support a sample rate of 12000.  So, `plughw:0,0` is usually required, but this is only shown with the `--list-pcms` option.  This option also shows the pulse device when available, as well as whatever has been defined with a `.asoundrc` file.  If you would like to work on changes to GetInputDeviceCollection() and GetOutputDeviceCollection() in src/linux/ALSA.c, that would be helpful.   If you restrict your changes to these two functions, it will not interfere with the other changes that I am currently working on in that same file.  While I'd like to see output similar to what arecord/aplay produce, the code to generate these lists must not be derived from the GPL licensed code used by those commands since that would be incompatible with the MIT license used for ardopcf.

The structure of `char **CaptureDevices` and `char **PlaybackDevices` with `int CaptureDevicesCound` and `int PlaybackDevicesCount` are already common between Waveform.c and ALSA.c to facilitate use with the CAPTUREDEVICES and PLAYBACKDEVICES host commands.  Currently, these structures are built for both systems, but are only really used in Waveform.c, where it is common for the user to specify audio devices using an index into these lists.  The host commands will need to build a single string from these arrays of strings.  Also, the Ardop specification currently says that the return of the CAPTUREDEVICES and PLAYBACKDEVICES host commands shall be a comma separated list.  It does not specify how to handle commas within device names.  This probably wasn't an issue with the original Windows implementation of Ardop and thus wasn't mentioned in the specification.  However, commas commonly appear in ALSA device names.  So, it will be necessary to explore how best to handle this.

radio-miskovice, 2025-03-04:
Peter, I forgot, as always, that I have to look into dev branch if I want to see what's up recently. I will checkout the development branch to better understand the recent changes and see what exactly I can do.

radio-miskovice, 2025-03-05:
P.S. there is a related issue (not formalized yet), and that is to allow ardopcf start even without any sound interface selected. The idea is to be able to initialize the modem completely from a host application. I feel it is independent of the current issue but it is not a priority before the current issue is resolved.
Last night I've done some research regarding device detection and would like to test it today if time allows. Will keep you posted. I may be unable to test all possible linux configurations, as all my available machines have PulseAudio.

radio-miskovice, 2025-03-06:
It works fine, I would like to have a talk as to how to integrate this.

pflarue, 2025-03-07:
How can I help?  I looked at https://github.com/radio-miskovice/ardop.  I don't see your changes, but perhaps I'm looking in the wrong place.  Please create a new branch there with your changes.  If those changes are ready to use, create a pull request to the develop branch under my account.  If you have questions, or want me to look at something before you create a pull request, put a link in this issue, and I'll do so.

I'll only have internet accesses intermittently for the next few weeks, so be patient with me and don't be surprised if it sometimes takes me a few days to respond.

radio-miskovice, 2025-03-15:
Hi, Peter, 

I will push the current status this weekend - 'issue-139-alt_dev_list_linux' branch in my fork. 

I got a bit stuck. I could retrieve device names and input/output type. Some have IOID null, some of these are virtual devices (PulseAudio plugins?) others are compound devices providing both input and output. 

I started studying documentation of the alsa project but it seems to be somewhat messy and without context, so it is difficult. 
Right now I would like to decide what to do next:
- learn how to match device name (actually sort identifier á la hw:CARD=PCH,DEV=0) with other alsa functions I can see in the original code. Especially when I have the "name" how can I use it to open a particular device. Secondary question is, if I have a device like Digirig, how do I pick the proper subdevice to be used as playback or capture for Ardopcf.
- maybe create a special structure (I tried to design one already, it's in the code) to hold both the ID/name and the description.

I think we should sync once a week to coordinate how to develop this part :)

radio-miskovice, 2025-03-15:
UPDATE:
I think I got through the alsa functions, see my "research sandbox" repo alsa-research-01.

Next step: 
1) I will write a new function, something like "GetDevicesCollections" to gather both playback and capture devices in one run. The basis is in my research sandbox, so I will basically add a few things, delete some debugging prints etc.

2) In order not to break the existing code, I will place the collections in new variables. The reason is that I want to prepare this for a change in the dependent functions, like opening a device or response to host commands CAPTUREDEVICE, CAPTUREDEVICES, PLAYBACKDEVICE, PLAYBACKDEVICES. That is something we will have to coordinate with you. The main reason is that I will use a different format that will allow for opening a device by its name (ID) and not the numbers, because the card and device numbers change every time I plug or unplug whatever rig or adapter (Digirig) with USB interface, while the "name", first, is constant, second, it gives at least a hint of what is connected (at least it reveals a USB thing).

3) I will write a device open function that will take card name as parameter. This will allow for consistently opening the intended device, not a random match of some volatile indexes, second, it may prevent ardopcf from unintentionally opening another device (the default), which is the "feature" of programs like VARA or JS8, which always make me freak out when something happens (such as unplug + plug USB rig or adapter) and they immediately reopen output to another device - HDMI display, headphones, main speakers, which is definitely what I don't want a program like JS8 to do. This stupid behavior is copycatted from behavior of the system for things like switching headphones/speakers when you unplug or replug while watching a movie or listening to music - not the use case for ARDOP, VARA, JS8, FT8, RTTY or whatever else of the ham radio kind of usage.

The new function willl have a different name and will be just added to the code, so that you can later on decide how to switch the methods.

pflarue, 2025-03-16:
Thank you for your effort.  I've copied your updates to my local computer, but have not yet looked at them.  Within the next couple of days I'll try to look at then and either provide comments here or integrate them with my recent changes.  

I just merged PRs #140.  This is the significant refactoring of the functions to open and configure ALSA audio devices that I have been working on for a while.  These changes, along with an updated version of GetDevicesCollections(), are useful steps toward resolving Issues #123 and #137.

radio-miskovice, 2025-03-16:
Hi, Peter, great that you had time to see it. I sometimes can't find enough time to look at these things as long as a whole week. Just two notes:

1. The name hint API returns names like "hw:CARD=PCH,DEV=7" and when this is used as parameter to snd_pcm_open() it works the same as things like "plughw:1,0". Not much, but it gives at least some basic information, especially when the user knows their system a little bit. OTOH on my chromebook that I use for digi, the card numbers are changing quite often as I plug and unplug Digirig or QMX and in many cases I need to investigate before I start ardopcf.

2. I did not include the code to discover channel count and sample rates. But it can be easily copied copy-paste from the existing functions, one block is the declarations and allocations, the other the actual data and display. I can addit in the meantime if you wish.

73 Jindra

pflarue, 2025-03-16:
Jindra,

I've created a new refactor-audio-devices-list branch (https://github.com/pflarue/ardop/tree/refactor-audio-devices-list) adapted from your work.  Thanks for your help.  I usually don't push my work to GitHub until it is ready to merge into the develop branch, but I wanted to let you see what I have done and see if you have any comments.  It isn't finished, but I think that it is close.  I also need to make some corresponding changes to the windows/Waveform.c implementation.  For now, in that branch, the new function runs and writes the results to the log, then ardopcf continues as it did before.

I'll probably leave it in this separate branch until I am ready to allow reconfiguration of audio devices with host commands.  That is still going to take some time since I'll need to make some architectural changes to allow ardopcf to operate when it does not have audio devices to use.  Currently, when that occurs ardopcf shuts down.  Preparing for that may suggest some additional changes including reformatting of the strings.  I also need to decide whether or not to add channel count and sample rate info.

armysarge, 2025-04-25:
Good day,

Just wanted to say how excited I am to be part of this project. I've always been interested in DSP and developing my own method, but I lacked the knowledge . Finding this open-source project has been a great opportunity, and I really appreciate the solid groundwork that’s already been laid.

I tend to learn best by diving head-first into complex challenges, so I’m looking forward to contributing and helping polish this project alongside the team.

On a more technical note – I’ve implemented changes to the CAPTURE and PLAYBACK commands, enabling users to set input/output devices without needing to restart the app. I’ve tested it on Windows and will be testing with ALSA on my Linux machine tonight. Feel free to check out my repo; I’ll be actively updating and refining ardopcf.

I’m also currently working on another modem project that explores various DSP techniques, with the end goal of creating a chat app similar to VarAC that would connect to this modem.

73,
ZS6SHZ

pflarue, 2025-04-25:
@armysarge,  

I am happy to learn of your interest in contributing to this project.  

However, the timing for the change you describe isn't great.  I've been working on a major update that sounds like it overlaps with the change you describe.  I believe that I'm very close to having it ready to push to GitHub.  The Linux version is working, and I'm cleaning up some minor remaining inconsistencies in the Windows version.  My change fully implements the CAPTURE, PLAYBACK, CAPTUREDEVICES, and PLAYBACKDEVICES host commands as well as the CODEC, RADIOPTT, and RADIOCTRLPORT host commands.  It also adds some new commands and cleanly handles hardware devices that provide audio plus any of the supported CAT/PTT control options being disconnected/turned off while in use followed by recovery when they (or alternative replacement devices) are connected/turned on.  This change also allows ardopcf to be started with none of these audio or CAT/PTT control devices configured via the command line, and thus it no longer tries to open any default audio devices if none are specified.

As I suspect that you have also seen, changing the hardware configuration while ardopcf is running, and continuing operation while no audio devices are open, requires some widespread structural changes.  So, even if we have achieved some of the same goals, I think that it is unlikely that our changes are fully compatible.

So, I'm going to wait until I push my changes before looking at how you have handled these issues.  Usually, when I push new changes to GitHub, I immediately merge them into the develop branch.  However, in this case I think that I'll leave them as a separate branch (off of develop) for a while.  This will allow us to compare our implementations and see whether some combination of our approaches might be better than what either of us have produced on our own.

I've also created a variant of host/python/diagnostichost.py that has allowed me to create semi-automated tests scripts to validate ardopcf's responses to host commands while prompting the user to do things such as turn devices on/off or connect/disconnect them.  Without this, testing correct behavior of changes to the audio configuration as well as the full range of CAT/PTT control devices was too cumbersome to manually do testing that was sufficiently comprehensive.  While ardopcf can detect whether CaptureAudio is actually working, it cannot tell whether output audio has actually been transmitted.  So, these new test scripts also depend on the user to confirm that audio output and PTT control are working correctly resulting in audio being heard on another radio during these tests.

I'll add another comment to this Issue when I push my new changes.

Peter LaRue
AI7YN

armysarge, 2025-04-25:
Thats great, looking forward to your merge then. My changes wasnt that
major, mostly worked in hostcommands, tested the commands using 2 ardopcf
processes running on virtual audio cables and it seemed to have worked
quite well. Didnt do comprehensive testing but I heard audio on both ends
and communications was successful.

Looking forward to your updates.
If there is anything specific you want me to look at and work, please so
let me know. Will glady try and take some load off of you.

pflarue, 2025-05-07:
New PR #152 of the interactive-configuration branch into the develop branch satisfies this Issue.  I encourage anyone interested in these changes to try it, and to look at the updated documentation (currently in the develop branch) which describes some of the new functionality.

New features include the ability to reconfigure most everything except for host/webgui port assignments and a few logging configuration options via host commands.  This includes the full ability to reconfigure audio and CAT/PTT devices via host commands and from the WebGui.  It is now possible to recover from a radio or radio interface being turned off or disconnected while in use.  See USAGE_linux.md#using-pulse-audio-devices for one example of when this fails (32-bit ARM with pulse audio device).

New status indicators in the WebGui are intended to help the user recognize when audio or CAT/PTT devices are not correctly configured.  Previously ardopcf would exit when devices could not be configured correctly, but this is no longer the case.

The equivalent of the CAPTUREDEVICES and PLAYBACKDEVICES host commands are used to populate audio device lists in the WebGUI for easy configuration.  Similar lists of CAT/PTT devices are not currently available.  So, it is necessary to type device names into inputs in the WebGui to reconfigure those.  I plan to look into whether cross-platform solutions can be developed to provide suitably comprehensive lists of possible CAT/PTT devices.  

This is a major rework that touches many parts of the ardopcf source code.  With so many changes, and many new features, new bugs have probably been introduced.  So, I encourage their use to help me find bugs that should be fixed before the next release.

Peter LaRue
AI7YN

closed 2025-05-07 by pflarue