pflarue/ardop #148: Alsa error on every TX -although it works ok

TAj-src, 2025-04-08:
`07:58:16.707 V A Broken Pipe error occured while attempting audio device write.  For some sound devices such as ALSA dmix and its derivatives, this occurs at the start of each transmission.  Attempting to recover.  The ALSA error in snd_pcm_recover() that follows is related to this.
07:58:16.708 V ALSA Error at pcm.c:8306 snd_pcm_recover()
`

TAj-src, 2025-04-11:
thanks for the information. Yes this is in full debug mode as I was tracing the ping problem and a bpq32 ardop disconnect on initial connection.I would leave this in for completeness. unless you add debug options to disable/enable modules, like alas. I would not bother.

craigerl, 2026-09-07:
#cd /tmp && /usr/local/bin/ardopcf --hostcommands "DRIVELEVEL 17" 8515 plughw:0,0 plughw:0,0 #audio error broken pipes

pure alsa, no pipewire/wireplumber/pulse,

2026-09-07T14:27:44.751702+01:00 digipi ardop.sh[1654]: [SendARQConnectRequest] MYCALL=KM6LYW  TARGET=KD6OAT bytPendingSessionID=a8
2026-09-07T14:27:44.751928+01:00 digipi ardop.sh[1654]: Sending Frame Type ConReq500M
2026-09-07T14:27:48.533731+01:00 digipi ardop.sh[1654]: [Repeating Last Frame]
2026-09-07T14:27:48.534017+01:00 digipi ardop.sh[1654]: Sending Frame Type ConReq500M
2026-09-07T14:27:48.534359+01:00 digipi ardop.sh[1654]: ALSA Error at pcm_hw.c:148 sync_ptr1(): Broken pipe
2026-09-07T14:27:52.314187+01:00 digipi ardop.sh[1654]: [Repeating Last Frame]
2026-09-07T14:27:52.314459+01:00 digipi ardop.sh[1654]: Sending Frame Type ConReq500M
2026-09-07T14:27:52.314722+01:00 digipi ardop.sh[1654]: ALSA Error at pcm_hw.c:148 sync_ptr1(): Broken pipe
2026-09-07T14:27:56.093341+01:00 digipi ardop.sh[1654]: [Repeating Last Frame]
2026-09-07T14:27:56.093618+01:00 digipi ardop.sh[1654]: Sending Frame Type ConReq500M
2026-09-07T14:27:56.093886+01:00 digipi ardop.sh[1654]: ALSA Error at pcm_hw.c:148 sync_ptr1(): Broken pipe

craigerl, 2026-09-07:
using "default" audio devices seems to fix the error  (vs plughw:0,0).   plughw is typically there to do rate matching, fails

pflarue, 2026-09-07:
@craigerl,

While I've discontinued development of ardopcf, I'm continuing to monitor posts to Issues.

I believe that this issue is primarily a symptom of an inappropriate audio system design for a program that must alternate between audio input (during RX) and output (during TX).  ardopcf is a purely single threaded program, even though audio on any modern general purpose OS inherently uses some sort of parallel processing.  So, for a  single threaded program like ardopcf, this means that the other part of the audio processing happens entirely in the OS/driver, which means the program has less control and visibility of what is happening.  A multithreaded program can move more audio processing into code that the program can more easily control and monitor.  While an Ardop implementation could do this, it would take a good bit of change to ardopcf to make it multithreaded.  A well designed multithreaded implementation of Ardop would be a great improvement over ardopcf.

This issue occurs because as currently structured, it is difficult to distinguish between a true error, and a condition that occurs repeatedly.  If, as originally described for this issue, the error message is occurring while operation is OK, then the easiest ways to eliminate annoying log messages are to either change the logging level or comment out the logging in the source code and recompile.  If a problem is disrupting operation, then it isn't what this issue was created to describe.