pflarue/ardop #156: Abnormal connection after hardware change

UgoPoddine, 2025-07-02:
Hello everyone,

I was using Ardop with pat on my PI3+ with PI/OS64 on my "old lady" fully analog rod for several years with joy, using for PTT an home made switch (practically an SSR on GPIO) and an home made very fast demon interrupt based for asserting GPIO.
I have upgraded my station hardware to a more powerful engine, running Debian, still with pat, but I was forced to adopt for PTT a commercial mechanical HID Relay on USB  and to setup rigctld for asserting PTT.
On the new machine the connection with my usual well running gateways is now impossible or very difficult (only one difficult connection accomplished), after the initial handshaking. In the attachment you can see the logs taken in very good propagation conditions.
I'm wondering an issue related to the PTT asserting delay due to rigctld + mechanical switch, but I'm not sure.
Can someone help ?

[ARDOPDebug8515_20250702.log](https://github.com/user-attachments/files/21011349/ARDOPDebug8515_20250702.log)

[patardop20250702.log](https://github.com/user-attachments/files/21011350/patardop20250702.log)

Thank-you
Best regards
73, IU1IPB Ugo

pflarue, 2025-07-27:
@UgoPoddine,  I've downloaded copies of your log files, but I don't have time today to study them thoroughly.  I'll try to do so and provide a better response in a few days.  It is possible that your hardware changes have introduced timing changes between PTT on/off and your transmitted audio.  If so, you may need to use the LEADER or TRAILER commands to make some adjustments.  These can be included in the command line via the -H or --hostcommands option.  Unfortunately, timing issues of this sort can be difficult to diagnose without also having access to log files from the station that you are trying to communicate with, and ideally that other station would be running ardopcf with verbose logging enabled.  If this isn't available, I may be able to suggest some changes you can try to see if they result in improvements.

pflarue, 2025-08-15:
@UgoPoddine, I'm sorry I did not respond sooner.  I have looked at your log file.  My initial guess was that the problem might have been related to audio/PTT latency, but this was wrong.  I now see that problem is that your receive audio was turned up too high.

In your log file I see several log messages that say "Your input signal is probably clipping.  If you see this message repeated in the next 20-30 seconds, turn down your RX input until this message stops repeating."  You need to follow this advice and turn down your RX audio level.  This can be done with alsamixer (or some equivalent Linux GUI tool) or via settings on your radio.