pflarue/ardop #158: Send2Tone doesn't have any logging output

askpatrickw, 2025-07-24:
SendID has logging outpub, but  Send2Tone doesn't have any logging output. The Send5SecTwoTone() function only calls GetTwoToneLeaderWithSync(250) and SoundFlush() without any logging, while SendID goes through the modulation functions in Modulate.c that include
  ZF_LOGI("Sending Frame Type %s", strType).

  This behavior is consistent across all platforms (Linux, Windows, macOS) since Send5SecTwoTone() is in the shared src/common/ARDOPC.c file.

  To match SendID's behavior, you could add a simple log message to Send5SecTwoTone():

  void Send5SecTwoTone()
  {
      ZF_LOGI("Sending two-tone test signal");
      initFilter(200, 1500);
      GetTwoToneLeaderWithSync(250);
      SoundFlush();
  }

  This would provide consistent user feedback for all transmission types.

pflarue, 2025-07-24:
I have added a lot of additional logging output (mostly at the DEBUG and VERBOSE levels) as I've been working to improve ardopcf.  I will gladly accept pull requests from anyone who feels that some action is not adequately reflected in the log, and that adding it will aid in observing and understanding of the program operation.

askpatrickw, 2025-07-26:
The reason I opened this is when I was testing the MacOS changes (which weren't working at the time), I was clicking the web buttons and I could see SendID would post to the log, but Send2Tone did nothing, so I wasn't sure if it was working or not at all. It'd be helpful if the WebUI buttons all generated a log entry.