pflarue/ardop #164: Windows build on Windows fails (develop branch, commit  443fb60)

radio-miskovice, 2025-07-31:
I could identify two issues in the Makefile but I am not exactly sure how to fix them because of my poor knowledge of make and Makefile features and uncertainty as to what exactly needs to be built and linked.

## Issue 1

Commands to generate C source code from HTML and Javascript files fail on Windows because directory separator in these commands are kept as forward slash (not substituted to backslash). E.g.

```
build/windows/lib/txt2c/txt2c webgui/webgui.html build/windows/src/common/gen-webgui.html.c webgui_html
```
should be 
```
build\windows\lib\txt2c\txt2c webgui\webgui.html build\windows\src\common\gen-webgui.html.c webgui_html
```

Side note: some Windows users could use Git Bash available in their installation of Git. I may try to add some conditions that could modify the make process if started on Windows but using Bash. This is detectable by checking SHELL environment variable: it contains /usr/bin/bash if running in Bash and does not exist in CMD.EXE. Similarly, there may be some more differences if the make is run from Powershell. (Oh, nightmare)

## Issue 2

The linker command refers file **build/windows/common/ardopcf.o**, but this file is not built (does not exist). **ardopcf.o** is not included in the **OBJS** variable and stands alone in **OBJS_EXE** as the only file. I will try to look why it is not compiled but cannot promise anything, being quite unfamiliar with the recent days make.

radio-miskovice, 2025-07-31:
I forgot to mention that the *mingw32-make.exe* actually does not compile anything, it just generates the following error:
```
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
mingw32-make: *** [makefile:224: build/windows/src/common/ardopcf.o] Error 2
```
And it is completely obscure why this happens. When I build from the *master* branch, everything runs as expected and I get _ardopcf.exe_ in matter of seconds.

However, _mingw32-make.exe_ can generate a command file by invoking it with **-n** parameter (dry run) and capturing the output into a comd file, like this:

```
mingw32-make.exe -n > build.cmd
```
which then builds something when run as command, except those two bugs described in this issue. The same is true for
```
mingw32-make.exe clean
```

pflarue, 2025-08-15:
@radio-miskovice,  I'm a bit confused by this Issue. 

I just retried building for Windows using both commit 443fb60 that you had problems with, and the current commit in the develop branch when I'm writing this of cb2c4c1.  Both work fine.  I'm using a mingw32 installation downloaded from https://winlibs.com/ as described in BUILDING.md.  I run mingw32-make from a Windows "Command Prompt" window.

I assume that this build of ming32-make is designed to handle the conversion of the path separators.  If not mingw32-make, then something else in this toolchain takes care of it.

I'm further confused by your error message referencing something in /usr/bin/sh.  That looks like a Linux error message rather than anything I'd expect from windows.  I suspect that this has something to do with git bash that you mentioned.  That isn't a tool that I use, and I suspect that it is the cause of your problems.  As also mentioned in BUILDING.md, I use git installed from a download at https://git-scm.com/downloads/win.

I suggest that you try these tools.  If that still doesn't work, I'll try to help you resolve any remaining problems.  If it does work, please close this Issue.